ros2/launch_ros #308: WaitForToppics spends a long time waiting for the thread to join

sloretz, 2022-04-07:
While debugging #304 I noticed `WaitForTopics` spends a lot of time joining the thread it creates (up to 1 second).

https://github.com/ros2/launch_ros/blob/a114e1fd5d92a0e672cab31002bc6e086521e985/launch_testing_ros/launch_testing_ros/wait_for_topics.py#L82

It looks like the thread spins for a whole second before checking if it should shutdown:

https://github.com/ros2/launch_ros/blob/a114e1fd5d92a0e672cab31002bc6e086521e985/launch_testing_ros/launch_testing_ros/wait_for_topics.py#L73-L74

I suspect this could be fixed by deleting `_spin_function` and using [`self.__ros_executor.spin`](https://github.com/ros2/rclpy/blob/8e8c978908f36e4cb5f34b3417eb6f912789f76d/rclpy/rclpy/executors.py#L276) as the thread target.
Then the thread could be joined must faster by shutting down the `rclpy.Context` before trying to join it.

Keng12, 2022-04-19:
Couldn't the same be achieved by changing the timeout parameter from 1.0 to 0? I. e.:

`self.__ros_executor.spin_once(0.0)`

sloretz, 2022-04-19:
> Couldn't the same be achieved by changing the timeout parameter from 1.0 to 0?

`0` means the executor [won't wait for work](https://github.com/ros2/rclpy/blob/6f7cfd0c73bda1afefba36b6785516f343d6b634/rclpy/rclpy/executors.py#L309-L310). Without any other blocking it would use up to a full CPU core.

Keng12, 2022-04-23:
I see. Then I will implement your suggestions and create a PR.

jacobperron, 2022-04-29:
Fixed in #314

closed 2022-04-29 by jacobperron