ros2/launch_ros #318: Action that automatically brings LifecycleNode to active state

ipa-cmh, 2022-04-29:
## Feature request

#### Feature description
Handling of LifecycleNodes and especially bringing them to active state is pretty lengthy when writing it in a launch file currently. This should be easier, especially for ease of debugging and development of lifecycle nodes. Preferably a one liner.

#### Implementation considerations
I can think of two implementation options:
* integrated into LifecycleNode action directly via parameter 
* a seperate action that inherits from LifecycleNode

A proof of concept implementation of the second option is in

jacobperron, 2022-05-02:
If I understand the request, you are looking for a shortcut to do the following:

```python
import launch
from launch.actions import EmitEvent

from launch_ros.actions import LifecycleNode
from launch_ros.events.lifecycle import ChangeState
from launch_ros.events.matchers import matches_node_name

import lifecycle_msgs.msg

def generate_launch_description():
    return launch.LaunchDescription([
        LifecycleNode(
            name='talker',
            namespace='',
            package='lifecycle',
            executable='lifecycle_talker',
            output='screen'
        ),  
        EmitEvent(
            event=ChangeState(
                lifecycle_node_matcher=matches_node_name('talker'),
                transition_id=lifecycle_msgs.msg.Transition.TRANSITION_CONFIGURE
            )   
        ),  
        EmitEvent(
            event=ChangeState(
                lifecycle_node_matcher=matches_node_name('talker'),
                transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE
            )   
        ),
    ])
```

Rather than introducing a special LifeycleNode action, it might be better to just provide some syntactic sugar to replace the two `EmitEvent` actions. Some options I can think of,

1. Add arguments to the existing `LifecycleNode` action to optionally transition to a different state:

```python
    return launch.LaunchDescription([
        LifecycleNode(
            name='talker',
            namespace='',
            package='lifecycle',
            executable='lifecycle_talker',
            output='screen',
            transition_to=lifecycle_msgs.msg.State.PRIMARY_STATE_ACTIVE
        ), 
    ])
```
2. Introduce a `TransitionLifecycle` action that simply emits the events you would like. It might look like this:

```python
def generate_launch_description():
    return launch.LaunchDescription([
        LifecycleNode(
            name='talker',
            namespace='',
            package='lifecycle',
            executable='lifecycle_talker',
            output='screen'
        ),  
        TransitionLifecycleNode(
            lifecycle_node_matcher=matches_node_name('talker'),
            transitions=[
                lifecycle_msgs.msg.Transition.TRANSITION_CONFIGURE, 
                lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE,
            ]
        ),  
    ])
```

The second options is more verbose but it has the benefit that we can transition multiple lifecycle nodes to the activated state with one action, by providing an appropriate matcher function.

ipa-cmh, 2022-05-03:
Regarding the problem description: I have not tested if just emitting events works in all cases. StateChange events are processed asynchronously by the LifecycleNode action, so if we have a node that takes some time to configure the activate event would probably be emitted before the node is in the inactive state and the transition to active state would fail.

I have been using the following code to get LifecycleNodes to active state (configure and then wait for inactive state event, then activate). This is a little more nasty to write an really blows up your launch file if you have multiple nodes. Actually copy pasting to other launchfile with other names has lead to some errors because of missing one or tow places that actually need to be changed...

```python
    talker = launch_ros.actions.LifecycleNode(
        name="talker", 
        namespace="", 
        package="lifecycle", 
        output="screen", 
        executable="lifecycle_talker",
    )
    talker_activate = launch.actions.RegisterEventHandler(
        launch_ros.event_handlers.OnStateTransition(
            target_lifecycle_node=talker, goal_state='inactive',
            entities=[
                launch.actions.EmitEvent(
                    event=launch_ros.events.lifecycle.ChangeState(
                        lifecycle_node_matcher=launch.events.matches_action(talker),
                        transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE
                    )
                ),
            ],
        )
    )
    talker_configure = launch.actions.EmitEvent(
        event=launch_ros.events.lifecycle.ChangeState(
            lifecycle_node_matcher=launch.events.matches_action(talker),
            transition_id=lifecycle_msgs.msg.Transition.TRANSITION_CONFIGURE,
        )
    )
    ld.add_action(talker_activate)
    ld.add_action(talker)
    ld.add_action(talker_configure)
```

Assuming this is the correct way of doing things:

* Option 1: Add arguments to the existing LifecycleNode action to optionally transition to a different state
   * We would probably need to implement an input checker to see whether the indicated state even makes sense. 
   * Probably restrict this to inactive and active state as a target, indicate not implemented for other targets.

* Option 2: Introduce a TransitionLifecycle action that simply emits the events you would like.
  * We would probabyl need to check for each transition, that it was performed (maybe without failure) before emitting the next StateChange event. 
  * The assumtion would be the user specifies a correct sequence of transitions. When sequence is infeasible or a transition fails, the action fails.
  * What should be the behaviour if a matcher matches multiple lifecycle nodes and the transitions fail on one lifecycle node? Do we throw an error and continue? Probably, we should fail the whole launch in this case, as the system cannot be put in the targetted state?

I would be in favor of **Option 2**, as it seperates lifecycle management from the node (action) itself.

methylDragon, 2022-12-14:
Closing this with the merge of https://github.com/ros2/launch_ros/pull/317

closed 2022-12-14 by methylDragon