ros2/launch_ros #319: Numbers in scientific notation are not tolerated

roncapat, 2022-06-07:
https://github.com/ros2/launch_ros/blob/476ff34b4152c56d5e6c8ce56bf18ccd109cc5dc/launch_ros/launch_ros/utilities/evaluate_parameters.py#L47-L58

This code rejects numeric lists containing numbers in scientific notation, like:
[1.0, 1e-9]

clalancette, 2022-06-07:
Are you sure about that?

```bash
$ python3 -c 'print(type(1e-9))'
<class 'float'>
```

Also, if I take that function, put it in a file, and then call it, like the following:
```python
def check_sequence_type_is_allowed(sequence):
    # Check if the items of the sequence aren't dissimilar.
    # Also, check if the item type is one of bool, int, float, str.
    if not sequence:  # Don't allow empty sequences
        return False
    subtype = type(sequence[0])
    if subtype not in (bool, int, float, str):
        return False
    for item in sequence:
        if not isinstance(item, subtype):
            return False
    return True

print('Empty:', check_sequence_type_is_allowed([]))
print('Float:', check_sequence_type_is_allowed([1.0]))
print('Scientific:', check_sequence_type_is_allowed([1.0, 1e-9]))
```

I get:
```bash
Empty: False
Float: True
Scientific: True
```

What's the larger problem that you are having?

roncapat, 2022-06-07:
The problem is in the ROS components parameter load chain. I think (but now I will give it a try) that scientific numbers are parsed from yaml config files as strings and not floats.

I encountered an error wile switching robot_localization package from being a node to being a component. Small covariances in config file (1e-9) started to throw the exception from evaluate_parameters.py

roncapat, 2022-06-07:
I printed `yaml_evaluated_value` variable prior to calling `check_sequence_type_is_allowed()` and got:

`[1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, '1e-9', 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, '1e-9', 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, '1e-9', 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, '1e-9', 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, '1e-9']`

closed 2022-06-07 by roncapat

roncapat, 2022-06-07:
Thanks for spotting the issue in the library! I am trying to figure it out (I work with C++ in ROS2 so I need a bit of time to figure out how to efficiently browse `rclpy` code) whether ROS2 rclpy Node implementation uses pyyaml or another implementation to read parameters from files.
Do you know more details about it?

roncapat, 2022-06-07:
I digged in rclpy and if you pass a filepath from launch file, rclpy seems to use the rcl implementation (so bypassing PyYAML).