ros2/launch_ros #328: LaunchConfiguration is not read from ComposableNode when included inside a group action

stevedanomodolor, 2022-10-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.1 LTS 
- Installation type:
  - source -> ros2  
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - ros2 rolling -> nav2
- Related issue 
 - https://github.com/ros-planning/navigation2/pull/3174

#### Steps to reproduce issue

```
ComposableNode(
                        condition=LaunchConfigurationNotEquals('map', ''),
                        package='nav2_map_server',
                        plugin='nav2_map_server::MapServer',
                        name='map_server',
                        parameters=[configured_params, {'yaml_filename': map_yaml_file}],
                        remappings=remappings
                        ),
ComposableNode(
                        condition=LaunchConfigurationNotEquals('map', ''),
                        package='nav2_map_server',
                        plugin='nav2_map_server::MapServer',
                        name='map_server',
                        parameters=[configured_params, {'yaml_filename': map_yaml_file}],
                        remappings=remappings
                        ),
```

#### Expected behavior
It should pay attention to the condition but executes both composable node. The issue #114 should have fixed it but seems not to have.

#### Actual behavior
It does not pay attention to the condition at all and executes both composablenode

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2022-10-12:
So we do have tests to ensure that `condition` works in a `ComposableNode` with `IfCondition` and `UnlessCondition`:

* https://github.com/ros2/launch_ros/blob/73098cfb2ef009b7db822ff9dee12504def814e2/test_launch_ros/test/test_launch_ros/actions/test_composable_node_container.py#L165
* https://github.com/ros2/launch_ros/blob/73098cfb2ef009b7db822ff9dee12504def814e2/test_launch_ros/test/test_launch_ros/actions/test_composable_node_container.py#L172

So I'm somewhat confident the basic functionality works.

That said, we don't have tests for `LaunchConfigurationNotEquals` with `ComposableNode`.  I don't even know if it is supposed to work.  Pinging @adityapande-1995 @jacobperron @hidmic for thoughts.

methylDragon, 2022-11-01:
The LaunchConfigurationEquals and LaunchConfigurationNotEquals conditions are deprecated.

Try with the EqualsSubstitution instead: https://github.com/ros2/launch/blob/rolling/launch/launch/conditions/launch_configuration_equals.py

stevedanomodolor, 2022-11-06:
How do i upgrade the launch package to have this changes, it seems that this feature was added not recently. I am using ros2 rolling. @methylDragon

methylDragon, 2022-11-07:
Try
```
sudo apt update
sudo apt upgrade --only-upgrade ros-rolling-launch
```

stevedanomodolor, 2022-11-07:
I get this when I try to upgrade, is this feature available in ros-rolling yet?
```
root@stevedan-OMEN-Laptop-15-ek0xxx:/opt/ros/rolling/lib/python3.10/site-packages/launch/substitutions# sudo apt-get install --only-upgrade ros-rolling-launch
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-rolling-launch is already the newest version (1.2.0-1jammy.20220914.023741).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@stevedan-OMEN-Laptop-15-ek0xxx:/opt/ros/rolling/lib/python3.10/site-packages/launch/substitutions# ls
__init__.py              environment_variable.py    python_expression.py
__pycache__              find_executable.py         substitution_failure.py
anon_name.py             launch_configuration.py    text_substitution.py
boolean_substitution.py  local_substitution.py      this_launch_file.py
command.py               path_join_substitution.py  this_launch_file_dir.py
root@stevedan-OMEN-Laptop-15-ek0xxx:/opt/ros/rolling/lib/python3.10/site-packages/launch/substitutions# 
```

methylDragon, 2022-11-07:
Hmm, you're right. The update was released 5 days ago, but it might take a little while for it to get through to the [main repos](http://repo.ros2.org/status_page/ros_rolling_ujv8.html). It should come out soon though.

hodnajit, 2022-12-02:
I am facing similar issue, but I am using IfCondition. You may find the issue description here:
https://github.com/ros2/launch_ros/issues/335

methylDragon, 2022-12-07:
> I am facing the similar issue, but I am using IfCondition. You may find the issue description here: #335

I think the issue is related. I am going to put some time into investigating this...

methylDragon, 2022-12-08:
Hello! This PR should address the issue (:
- https://github.com/ros2/launch_ros/pull/339

@stevedanomodolor

stevedanomodolor, 2022-12-08:
Thanks for the update

SteveMacenski, 2023-03-03:
Changes being made now in Nav2 https://github.com/ros-planning/navigation2/pull/3445/files