ros2/launch_ros #335: IfCondition is ignored in ComposableNode

hodnajit, 2022-12-02:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-rolling-launch (1.3.0-1jammy.20221102.202147)
- DDS implementation:
  - rmw_cyclonedds_cpp
- Client library (if applicable):
  - launch

#### Steps to reproduce issue

Run example launchfile.

```
#!/usr/bin/env python3
#
# Copyright (C) 2020 DataVision s.r.o. All rights reserved.

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.conditions import IfCondition
from launch.actions import DeclareLaunchArgument, GroupAction
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import LoadComposableNodes
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    show_image = LaunchConfiguration('show_image')

    declare_show_image_cmd = DeclareLaunchArgument(
        'show_image', default_value='False',
        description='Show image if True')

    use_composition = LaunchConfiguration('use_composition')

    declare_use_composition_cmd = DeclareLaunchArgument(
        'use_composition', default_value='True',
        description='Launch GroupAction if True')

    container = Node(
        condition=IfCondition(use_composition),
        name='image_container',
        package='rclcpp_components',
        executable='component_container',
        output='both',
    )

    load_composable_nodes = LoadComposableNodes(
        condition=IfCondition(use_composition),
        target_container='image_container',
        composable_node_descriptions=[
            ComposableNode(
                package='image_tools',
                plugin='image_tools::Cam2Image',
                name='cam2image',
                remappings=[('/image', '/burgerimage')],
                parameters=[{'width': 320, 'height': 240, 'burger_mode': True, 'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True}],
            ),
            ComposableNode(
                package='image_tools',
                plugin='image_tools::ShowImage',
                name='showimage',
                condition=IfCondition(show_image),
                remappings=[('/image', '/burgerimage')],
                parameters=[{'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True},]
            ),
        ],
    )

    # Create the launch description and populate
    ld = LaunchDescription()

    # Declare the launch options
    ld.add_action(declare_show_image_cmd)
    ld.add_action(declare_use_composition_cmd)

    # Add composable container
    ld.add_action(container)

    # Add the actions to launch of the nodes
    ld.add_action(load_composable_nodes)
    return ld

```

#### Expected behavior
The node /showimage is not launched.

#### Actual behavior
The node /showimage is launched.

hodnajit, 2022-12-02:
Maybe better example (similar to this test) https://github.com/ros2/launch_ros/blob/73098cfb2ef009b7db822ff9dee12504def814e2/test_launch_ros/test/test_launch_ros/actions/test_composable_node_container.py#L172

`#!/usr/bin/env python3
#
# Copyright (C) 2020 DataVision s.r.o. All rights reserved.

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.conditions import IfCondition, UnlessCondition
from launch.actions import DeclareLaunchArgument, GroupAction
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import LoadComposableNodes, ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

TEST_CONTAINER_NAME = 'test_component_container_node_name'
TEST_NODE_NAME = 'test_composable_node_name'
TEST_NODE_NAMESPACE = 'test_composable_node_namespace'

def generate_launch_description():
    flag = LaunchConfiguration('flag')

    TEST_NODE_NAME_1 = 'test_component_container_node_name_1'
    TEST_NODE_NAME_2 = 'test_component_container_node_name_2'

    declare_flag_cmd = DeclareLaunchArgument(name='flag', default_value='False')

    container = Node(
        name=TEST_CONTAINER_NAME,
        package='rclcpp_components',
        executable='component_container',
        output='both',
    )

    load_composable_nodes = LoadComposableNodes(
        target_container=TEST_CONTAINER_NAME,
        composable_node_descriptions=[
            ComposableNode(
                package='composition',
                plugin='composition::Listener',
                name=TEST_NODE_NAME_1,
                condition=UnlessCondition(flag)
            ),
            ComposableNode(
                package='composition',
                plugin='composition::Listener',
                name=TEST_NODE_NAME_2,
                condition=IfCondition(flag)
            )
        ],
    )

    # Create the launch description and populate
    ld = LaunchDescription()

    # Declare the launch options
    ld.add_action(declare_flag_cmd)

    # Add composable container
    ld.add_action(container)

    # Add the actions to launch of the nodes
    ld.add_action(load_composable_nodes)

    return ld
`

methylDragon, 2022-12-07:
Reformatting for convenience

```
#!/usr/bin/env python3

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.conditions import IfCondition, UnlessCondition
from launch.actions import DeclareLaunchArgument, GroupAction
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import LoadComposableNodes, ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

TEST_CONTAINER_NAME = 'test_component_container_node_name'
TEST_NODE_NAME = 'test_composable_node_name'
TEST_NODE_NAMESPACE = 'test_composable_node_namespace'

def generate_launch_description():
flag = LaunchConfiguration('flag')

TEST_NODE_NAME_1 = 'test_component_container_node_name_1'
TEST_NODE_NAME_2 = 'test_component_container_node_name_2'

declare_flag_cmd = DeclareLaunchArgument(name='flag', default_value='False')

container = Node(
    name=TEST_CONTAINER_NAME,
    package='rclcpp_components',
    executable='component_container',
    output='both',
)

load_composable_nodes = LoadComposableNodes(
    target_container=TEST_CONTAINER_NAME,
    composable_node_descriptions=[
        ComposableNode(
            package='composition',
            plugin='composition::Listener',
            name=TEST_NODE_NAME_1,
            condition=UnlessCondition(flag)
        ),
        ComposableNode(
            package='composition',
            plugin='composition::Listener',
            name=TEST_NODE_NAME_2,
            condition=IfCondition(flag)
        )
    ],
)

# Create the launch description and populate
ld = LaunchDescription()

# Declare the launch options
ld.add_action(declare_flag_cmd)

# Add composable container
ld.add_action(container)

# Add the actions to launch of the nodes
ld.add_action(load_composable_nodes)

return ld
```

methylDragon, 2022-12-07:
Gotcha, I can reproduce this issue, now to look into it...

methylDragon, 2022-12-08:
Hello! This PR should address the issue (:

@hodnajit

hodnajit, 2022-12-08:
Thank you :) great job!

closed 2022-12-08 by hodnajit