ros2/launch_ros #346: Get content of topic from `launch_testing_ros::WaitForTopics`

LastStarDust, 2023-01-25:
## Feature request

#### Feature description
In the `WaitForTopics` sample code [here](https://programtalk.com/vs4/python/ros2/launch_ros/launch_testing_ros/test/examples/wait_for_topic_launch_test.py/), I see how I can easily test that a node has published a topic with a certain name, but it is not shown how to test the actual content of the topic.

It would be very helpful if the content of the topic would be available to the tester, too.

#### Implementation considerations
Looking deeper at the source code [here](https://github.com/ros2/launch_ros/blob/35b2ca9f99f193bab82f02141300c6102e0bdb71/launch_testing_ros/launch_testing_ros/wait_for_topics.py#L129), it seems that only the topic name is stored.

Because of Python duck-typing it should be more or less easy to just return the content of the topic. Something like
```python
    def callback_template(self, topic_name):

        def topic_callback(data):
            if topic_name not in self.received_topics:
                self.get_logger().debug('Message received for ' + topic_name)
                self.received_topics.add(topic_name)
                self.received_messages[topic_name].append(data)
            if self.received_topics == self.expected_topics:
                self.msg_event_object.set()

        return topic_callback
```
where `self.received_messages` is a dictionary where the key is the topic name and the value is the list of received messages. Just as an example.

LastStarDust, 2023-01-25:
If you deem this feature useful, I would be available to implement it and make a pull request.

adityapande-1995, 2023-02-10:
Hi @LastStarDust ! That is a good suggestion. While ``WaitForTopics`` was originally meant to just block until all the mentioned topics receive messages, it can be useful to know what the message was. A couple of questions : 

1. Should the message stored be the last message before ``WaitForTopics()`` returns, or the first message a topic receives ?
2. Should this workflow be preferred over adding a separate subscriber ?

If you could open a PR, I'll be happy to review it.

LastStarDust, 2023-02-13:
@adityapande-1995 Hello. Thank you for your feedback. I created a pull request #353 for this feature as you suggested.

Ryanf55, 2024-01-16:
Hello, I would like to re-iterate how important this feature is for creating production-ready ROS code that is tested at the middleware level.

In following the V-model of development in commercial projects that use ROS 2, it is very important to have tooling to send and receive messages, services, and actions to validate that the behavior of a node is in line with the requirements, and to be able to have **reliable** automated tests that can prove adherence to the requirements. 

Asserting that the node published a message, but ignoring the content is not sufficient enough of a test to demonstrate adherence to design requirements for any asynchronous publishers. 

A great example is to check that the header timestamp is not zero and is the same as the event that triggered the publish. 

Currently, my team is maintaining a bunch a layer on top of launch_ros to perform this kind of testing, and would much prefer to work with the ROS community to improve test infrastructure.

Ryanf55, 2024-01-20:
This is now backported to all active distros. It is now solved. Please close the issue as completed.

clalancette, 2024-01-21:
Closing as solved by #353

closed 2024-01-21 by clalancette