ros2/launch_ros #366: FeatureRequest: Add type hints to launch_testing_ros examples.

lexi-brt, 2023-06-20:
## Feature request

#### Feature description
It would be nice if the examples for launch_testing_ros had type hints since it would be easier to read and understand the example. This is a request to add those type hints.

For example it's unclear what the talker and proc_output parameter types are here: https://github.com/ros2/launch_ros/blob/rolling/launch_testing_ros/test/examples/talker_listener_launch_test.py#L89

clalancette, 2023-06-21:
Yes, agreed that would be nice.  I'll mark this as a "good first issue" and "help wanted".

Relativiteit, 2023-08-10:
@lexi-brt and @clalancette  I would like to make this one of my first contributions!

clalancette, 2023-08-10:
> @lexi-brt and @clalancette I would like to make this one of my first contributions!

Please go ahead and make the changes and open a pull request.  We're happy to review.

yaswanth1701, 2023-11-18:
Hi @clalancette , I don't seen any PRs for this particular issue. So,can I work on this issue?

clalancette, 2023-11-20:
> Hi @clalancette , I don't see any PRs for this particular issue. So,can I work on this issue?

Please go ahead and make the changes and open a pull request.  We are happy to review.

closed 2023-11-21 by clalancette