ros2/launch_ros #374: Simpler substitution for package files

jdlangs, 2023-08-15:
## Feature request

#### Feature description
Looking at a [launch file tutorial](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Using-Substitutions.html#parent-launch-file), I was struck by the verbose use of both the `PathJoinSubstitution` and `FindPackageShare` substitutions. It would be very easy to simplify this for the end-user with a single `FindPackageFile` substitution, for example. Is there any reason this wasn't done originally and would such a change be welcomed?

clalancette, 2023-08-24:
I don't think there is a specific reason this wasn't originally done.  We'd be happy to review a pull request that adds this functionality.

GlowingScrewdriver, 2023-12-21:
I've got two questions:

1. Would this substitution be something that takes in just a package name and a launch file name? As in:
```python3
PythonLaunchDescriptionSource([
   FindPackageFile ('launch_tutorial', 'example_substitutions.launch.py')
])
```

2. How appropriate would it be to have a similar feature for Yaml launch files? Just enough capability to allow importing a given launch file from a given package? It would definitely allow for a lot more launch files to be written in Yaml instead of Python.