ros2/launch_ros #380: Consider providing user control of log file base names

sea-bass, 2023-09-29:
## Feature request

#### Feature description
I have reported an issue in the `ros2cli` repo recently (https://github.com/ros2/ros2cli/issues/856) which hopefully will gain traction.

The general idea in the above issue -- and please read that for more details -- is that the default log file names right now are prepended with the executable name (in the C++ case) or the Python executable name (e.g., `python3`) in the Python case.

The investigation is leading to maybe providing a user-configurable parameter for `ros2 run` that allows the user to provide their own base name for log files.

Per @fujitatomoya's suggestion, I wanted to open this up here as well to ensure that the `launch_ros` tools track the above issue and have a way to expose this through `ros2 launch`, and the launch APIs, as well.

Tanishq30052002, 2025-04-15:
@sea-bass @fujitatomoya 
I found this issue from open robotics post on linedin, is this issue still needs to be solved? If yes can i take it up?

fujitatomoya, 2025-04-15:
@Tanishq30052002 you can take it! thanks!

kscottz, 2025-04-17:
@Tanishq30052002 how is it going? Make any progress?

Tanishq30052002, 2025-04-19:
@fujitatomoya https://github.com/ros2/rcl/pull/1127/files

i found a mr which was merged earlier and it allows user to add custom name for a log

so can you brief what exactly the purpose for this pr might be?

Tanishq30052002, 2025-04-19:
@fujitatomoya https://github.com/ros2/rcl/pull/1127
Can you please provide a brief of what is required, as I found a PR where we use the log file name as input. So what is the difference in this PR.
Thanks

Tanishq30052002, 2025-04-19:
@kscottz @fujitatomoya can you please help me in knowing how to process the args after taking them as input from the args parser?
I have been trying to figure it out but got stuck. Help is appreciated.

Tanishq30052002, 2025-04-19:
https://github.com/ros2/rcl/pull/1127

Also, I have gone through the previous PR, which helps to provide custom names in case of ros2 run.

Tanishq30052002, 2025-04-20:
I have implemented the structure such that user can set the name from cli and it will the the final log for launch file with that name

Tanishq30052002, 2025-04-20:
[image]
I hope this is the expected output, if I haven't misinterpreted the problem.

kscottz, 2025-04-21:
@fujitatomoya can you look at #461? 

@Tanishq30052002 protip: if you write `closes #<issue number>` in your PR it will automatically ping this thread. 😄 

I made small suggestion to fix your linter error. You should be able to click the accept suggestion button and run it through CI again.

fujitatomoya, 2025-04-21:
@sea-bass https://github.com/ros2/launch_ros/issues/380#issuecomment-2816931052 is what you request here, correct? for each executable node, we can use ``