ros2/launch_ros #398: feat: Adding a prefix option to node launch

wkaisertexas, 2024-06-13:
## Feature request

#### Feature description

Adding an extra parameter to the `Node` class to launch and executable with a specified prefix.

https://github.com/ros2/launch_ros/blob/8751ff4cd9060157831dcaeecd89611932aec608/launch_ros/launch_ros/actions/node.py#L112-L240

From what I can tell, this was a feature in `ros` which allowed a `prefix` to be specified which is launched ahead of the executable.

- [Ros Forums on Ros Launch and it use with GDB](https://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)

#### Implementation considerations

Potentially this feature could also be implemented with `ComposableNodes` API, however implementation and the use case there are a bit more vague.

Currently, I am using `Node(executable=f"{prefix} {executable}")` as a workaround, but this does not seem ideal.

clalancette, 2024-06-13:
I think we have this already in Launch; see http://docs.ros.org/en/jazzy/How-To-Guides/Getting-Backtraces-in-ROS-2.html#from-a-launch-file for an example.

Is that what you are asking about?

wkaisertexas, 2024-06-13:
Yes, I was a bit confused because it was lumped in as kwargs. Should there be a note about prefixing in the documentation for this method?

closed 2024-06-13 by wkaisertexas

clalancette, 2024-06-13:
> Yes, I was a bit confused because it was lumped in as kwargs. Should there be a note about prefixing in the documentation for this method?

Yep, that would be great!  Feel free to open a PR doing that, we are happy to review it.

wkaisertexas, 2024-06-13:
@clalancette I am not sure if this is the correct way to do this with sphinx and the documentation system