ros2/launch_ros #407: :woman_farmer: `RuntimeError: Context.init() must only be called once` test regression

claraberendsen, 2024-07-19:
## Bug report
It seems like #402 might have introduced a regression on the tests for all repeated jobs. 

[test_launch_ros.test.test_launch_ros.actions.test_ros_timer.test_timer_uses_sim_time[2-3]](https://ci.ros2.org/job/nightly_linux-rhel_repeated/1969/testReport/junit/test_launch_ros.test.test_launch_ros.actions/test_ros_timer/test_timer_uses_sim_time_2_3_/)
[test_launch_ros.test.test_launch_ros.actions.test_ros_timer.test_timer_uses_sim_time[3-3]](https://ci.ros2.org/job/nightly_linux-rhel_repeated/1969/testReport/junit/test_launch_ros.test.test_launch_ros.actions/test_ros_timer/test_timer_uses_sim_time_3_3_/)

The error message is: 

```
failed on setup with "RuntimeError: Context.init() must only be called once"
```
Previous example build without this error: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/2812/ 
Affected builds: 
[nightly_linux-rhel_repeated#1968](https://ci.ros2.org/job/nightly_linux-rhel_repeated/1968)
[nightly_linux_repeated#3521](https://ci.ros2.org/job/nightly_linux_repeated/3521)
[nightly_linux-aarch64_repeated#2823](https://ci.ros2.org/job/nightly_linux-aarch64_repeated/2823)

**Required Info:**

- Operating System:
  - Ubuntu
  - RHEL
- Version or commit hash:
  - 58564825f62730926f3609be5001e7a1cd145a23

#### Steps to reproduce issue
Run a build of any of the projects above and see the test fail

claraberendsen, 2024-07-19:
It seems to be related to a change in rclpy https://github.com/ros2/rclpy/pull/1298

clalancette, 2024-07-22:
This should be fixed by https://github.com/ros2/rclpy/pull/1314

clalancette, 2024-07-22:
Closing now that https://github.com/ros2/rclpy/pull/1314 has been merged.

closed 2024-07-22 by clalancette