ros2/launch_ros #408: :farmer: check_node_launch_test failing consistently in Iron and Humble Connext

Crola1702, 2024-07-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Jammy
- Installation type:
  - Source
- Version or commit hash:
  - Iron and Humble
- DDS implementation:
  - RTI Connext

#### Steps to reproduce issue
1. Run a build in [Humble](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/) or [Iron](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/) connect
2 See check_node_launch_test fail

#### Additional information

Reference build: 
* [Ici__nightly-connext_ubuntu_jammy_amd64#151](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/151) 
* [Hci__nightly-connext_ubuntu_jammy_amd64#258](https://build.ros2.org/job/Hci__nightly-connext_ubuntu_jammy_amd64/258)

Test regression:
* [launch_testing_ros.test.examples.check_node_launch_test.check_node_launch_test](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/151/testReport/junit/launch_testing_ros.test.examples/check_node_launch_test/check_node_launch_test/)

<details>
<summary>
Log output:
</summary>

```
=================================== FAILURES ===================================
_____________________ launch tests: check_node_launch_test _____________________

======================================================================================================================================================================================================
FAIL: check_node_launch_test.TestFixture.test_node_start
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/check_node_launch_test.py", line 56, in test_node_start
    assert node.wait_for_node('demo_node_1', 8.0), 'Node not found !'
AssertionError: Node not found !
assert False
 +  where False = <bound method MakeTestNode.wait_for_node of <check_node_launch_test.MakeTestNode object at 0x7facd297a770>>('demo_node_1', 8.0)
 +    where <bound method MakeTestNode.wait_for_node of <check_node_launch_test.MakeTestNode object at 0x7facd297a770>> = <check_node_launch_test.MakeTestNode object at 0x7facd297a770>.wait_for_node
----------------------------- Captured stdout call -----------------------------
[INFO] [launch]: All log files can be found below /home/buildfarm/.ros/log/2024-07-28-02-14-25-762362-2ab081e793eb-417268
[INFO] [launch]: Default logging verbosity is set to INFO
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
Waiting for node...
[INFO] [python3-2]: process started with pid [417328]
[python3-2] [INFO] [1722161673.935245860] [demo_node_1]: Publishing: "Hello World: 0"
[python3-2] [INFO] [1722161674.918469191] [demo_node_1]: Publishing: "Hello World: 1"
[INFO] [python3-2]: sending signal 'SIGINT' to process[python3-2]
[python3-2] Traceback (most recent call last):
[python3-2]   File "/tmp/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/talker.py", line 52, in <module>
[python3-2]     main()
[python3-2]   File "/tmp/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/talker.py", line 48, in main
[python3-2]     rclpy.shutdown()
[python3-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
[python3-2]     _shutdown(context=context)
[python3-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
[python3-2]     context.shutdown()
[python3-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
[python3-2]     self.__context.shutdown()
[python3-2] rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at /tmp/ws/src/ros2/rcl/rcl/src/rcl/init.c:333
[ERROR] [python3-2]: process has died [pid 417328, exit code 1, cmd '/usr/bin/python3 /tmp/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/talker.py --ros-args -r __node:=demo_node_1'].
----------------------------- Captured stderr call -----------------------------
test_node_start (check_node_launch_test.TestFixture) ... FAIL

======================================================================
FAIL: test_node_start (check_node_launch_test.TestFixture)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/src/ros2/launch_ros/launch_testing_ros/test/examples/check_node_launch_test.py", line 56, in test_node_start
    assert node.wait_for_node('demo_node_1', 8.0), 'Node not found !'
AssertionError: Node not found !
assert False
 +  where False = <bound method MakeTestNode.wait_for_node of <check_node_launch_test.MakeTestNode object at 0x7facd377a8f0>>('demo_node_1', 8.0)
 +    where <bound method MakeTestNode.wait_for_node of <check_node_launch_test.MakeTestNode object at 0x7facd377a8f0>> = <check_node_launch_test.MakeTestNode object at 0x7facd377a8f0>.wait_for_node

----------------------------------------------------------------------
Ran 1 test in 8.316s

FAILED (failures=1)
```

</details>

mjcarroll, 2024-08-08:
Maintenance triage meeting elected to backlog this.  This falls in the lowest priority set of CI jobs, so it is unlikely to get looked at in the near future.

Blast545, 2024-10-15:
It seems this is failing both in `launch_testing` and in `launch_testing_examples`, I will disable the test in both cases for RTI connext. Looks like repeated tests to me, see:

https://github.com/ros2/launch_ros/blob/iron/launch_testing_ros/test/examples/check_node_launch_test.py#L52

and

https://github.com/ros2/examples/blob/iron/launch_testing/launch_testing_examples/launch_testing_examples/check_node_launch_test.py#L55

Crola1702, 2024-11-12:
This issue was fixed