ros2/launch_ros #418: Create option for automatic bringup of lifecycle nodes in launch_ros API

SteveMacenski, 2024-10-08:
## Feature request

#### Feature description

Currently, transitioning up a lifecycle node can be done via an external node, lifecycle manager, or using Opaque functions with some rather nasty looking boilerplate with events. The event-based solution only works for `LifecycleNode` as that has the implementations for getting the states required. However, a `ComponentNode` can also be a lifecycle node and there is no way to my knowledge to perform lifecycle operations in launch_ros directly on the `ComponentNode` object. 

It would be nice if there was a `LifecycleNode` and `ComponentNode` field for `active_on_bringup` (or eq.) that incorporated this boilerplate to make this easier as many users of lifecycle nodes don't truly care to transition up and control the lifecycle of all their programs with such granularity.   

#### Implementation considerations

I don't think there are any unique implementation considerations.

mjcarroll, 2024-10-09:
This would be an excellent addition.  Do you have any bandwidth to work on it right now?  Alternatively does it make sense to discuss at the PMC meeting?

SteveMacenski, 2024-10-11:
I’m already traveling for ROSCon / ROSI and board meetings, so I won’t be able to work on this for at least a month. I’m reporting as a ‘nice to have’ enhancement that’s come up in discussions in the Nav2 community enough times now I figure it has general community interest. 

Maybe in December I could take a look, but if someone else has interest and capacity, I’m happy to hand it off.

SteveMacenski, 2024-12-19:
@mjcarroll I'm looking at this again and could use some advise as to how to go about it.

I'm seeing that the lifecycle node creates a service client and creates an event handler to change on transition when an event is emitted [[1]](https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/actions/lifecycle_node.py#L156-L163). I'm drawing that this is what is called when I do something [like this](https://github.com/ros-drivers/ros2_ouster_drivers/blob/ros2/ros2_ouster/launch/driver_launch.py#L55-L60) and using the end-user `EmitEvent` API. 

Well there's this neat client already located conveniently inside of `lifecycle_node.py` that seems begging to be used, but the question is about the best way in the maintainer's view:
- I can EmitEvent of the lifecycle transition and update the return to include those actions as things to execute 
- I can directly just use the `__rclpy_change_state_client` to call the transitions we're looking for in an auto-activate feature 
- I can possibly imagine using the feature in LifecycleTransition to use the known Node's name and do the 2 transitions for inactive and active through configure and activate in order

Which (if any) do you think is the right move? 

[1] https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/actions/lifecycle_node.py#L156-L163
[2] https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/actions/lifecycle_transition.py

---

More journaling: the next thing that I need to figure out is how to add and handle the `autostart` field in the `Node` base class (Note:  `ComponentNode` and `LifecycleNode` also is able to use them; since both are also able to be lifecycle nodes) as an option alongside the node's name, namespace, remappings, and so on so that we can actually trigger the transitions when requested. I know this is probably obvious to maintainers, but very much not so at first glance as a first-time contributor. 

As far as I can tell:
1. Add in the `autostart` field in the contructor arguments [[reference](https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/actions/node.py#L127)]
2. Set its value as a class member `self.__autostart`
3. Populate kwargs if its set [[reference](https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/actions/node.py#L310-L312)]
4. Create a getter function for getting the autostart member's value
5. Use the getter function inside of LifecycleNode.execute()` to enact the behavior discussed in the previous section 
6. Do ... something (?) ... inside of `_perform_substitutions` --> though I'm thinking probably not unless I'm missing something key

SteveMacenski, 2024-12-19:
Journaling a bit at this point (leaving breadcrumbs should I or someone else need to do something like this again):

Adding in the new field was easy, from the bullet list 1-4. That was solved / as straight forward as I expected.

I'm working on adding the autotransition using `LifecycleTransition`, but running into a few stumbles on circular dependencies that are in the works currently. I expect to have a PR open in the next couple of days

SteveMacenski, 2024-12-20:
PR #430 implements for `Node` and `LifecycleNode`. Composition node is TBD

SteveMacenski, 2024-12-20:
Continuing this afternoon - without feedback I'm just going to use my best judgement. I'm going with a lifecycle event manager utility that is used in the lifecycle node and component node :+1:

SteveMacenski, 2025-01-02:
The PR is fully updated and ready for review

PR #430 (2025-02-19): Autostarting lifecycle nodes and example launch file demo
This resolves #418 which implements autostarting lifecycle nodes. This is complete and ready for a review 

You'll notice a couple of key changes worth explaining:

1. There is a new `is_lifecycle_node` attribute of the node and lifecycle nodes which is needed to remove a circular dependency on the LifecycleNode within the `LifecycleTransition` class which only is type checking. I replace this type check with checking if a non-None object (1) has the attribute at all and (2) has a lifecycle attribute with appropriate logging between the [... cut]
PR #438 (2025-03-25): Autostarting lifecycle nodes and example launch file demo (backport #430)
This resolves #418 which implements autostarting lifecycle nodes. This is complete and ready for a review 

You'll notice a couple of key changes worth explaining:

1. There is a new `is_lifecycle_node` attribute of the node and lifecycle nodes which is needed to remove a circular dependency on the LifecycleNode within the `LifecycleTransition` class which only is type checking. I replace this type check with checking if a non-None object (1) has the attribute at all and (2) has a lifecycle attribute with appropriate log [... cut]