ros2/launch_ros #428: Namespace doesn't get applied to target container when using load_composable_node tag

mbharatheesha, 2024-12-04:
So, I have an xml launch (let's say: `pkg.launch.xml`) from binaries as follows:

<?xml version="1.0"?>
<launch>
    <arg name="some_arg" default="true"/>
   
    <node pkg="rclcpp_components" exec="component_container" name="container" >
    </node>
    <load_composable_node if="$(var some_arg)" target="container">
        <composable_node  pkg="pkg" plugin="pkg::SomePlugin" name="c_node" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>   

    <load_composable_node unless="$(var some_arg)" target="container">
        <composable_node  pkg="pkg" plugin="pkg::SomeOtherPlugin" name="c_node" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>
  
    <load_composable_node target="container">
        <composable_node  pkg="pkg" plugin="pkg::Plugin" name="c_node1" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>
</launch>

And, I have a top level launch file `top_level.launch.xml` as follows:

<launch>
    <arg name="my_arg" default="my_default" />

    <group>
    <push_ros_namespace namespace="$(var my_arg)" />
        <include file="$(find-pkg-share pkg)/launch/pkg.launch.xml" >
            <arg name="some_arg" value="false" />
        </include>
    </group>
</launch>

- Operating System:
  - ubuntu 24.04 LTS
- Installation type:
  - ROS2 Jazzy from binaries
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
`
ros2 launch top_level.launch.xml
`
#### Expected behavior
I expect that all composable nodes should get loaded into `my_default/container`

#### Actual behavior
The composable nodes seem to be looking for `container` instead

#### Additional information
The only way to get the composable nodes loaded into the namespaced container is to accept the namespace string as a launch argument and append it to the `target` tag, like so:
<?xml version="1.0"?>
<launch>
    <arg name="some_arg" default="true"/>
    <arg namespace="namespace" default="my_ns"/>

   
    <node pkg="rclcpp_components" exec="component_container" name="container" >
    </node>
    <load_composable_node if="$(var some_arg)" target="$(var namespace)/container">
        <composable_node  pkg="pkg" plugin="pkg::SomePlugin" name="c_node" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>   

    <load_composable_node unless="$(var some_arg)" target="$(var namespace)/container">
        <composable_node  pkg="pkg" plugin="pkg::SomeOtherPlugin" name="c_node" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>
  
    <load_composable_node target="$(var namespace)/container">
        <composable_node  pkg="pkg" plugin="pkg::Plugin" name="c_node1" namespace="">
            <param name="param" value="value" />
        </composable_node>
    </load_composable_node>
</launch>

And modify, the `top_level.launch.xml` like so:

<launch>
    <arg name="my_arg" default="my_default" />

    <group>
    <push_ros_namespace namespace="$(var my_arg)" />
        <include file="$(find-pkg-share pkg)/launch/pkg.launch.xml" >
            <arg name="some_arg" value="false" />
            <arg namespace="$(var namespace)" />
        </include>
    </group>
</launch>

But, this would only work, if the binary installed xml launch file has already considered the option to append namespaces to target container names. 

----
## Feature request
Enable the target containers in `load_composable_node` tags to acquire the namespace from the `push_ros_namespace` tag, if one is specified.

#### Note:

I am not sure if this is a bug if there's a different way the `<load_composable_node>` tag is to be used when using namespaces.

sloretz, 2024-12-13:
> I am not sure if this is a bug

Seems like a bug 🧇

mbharatheesha, 2024-12-13:
> > I am not sure if this is a bug
> 
> Seems like a bug 🧇

Alright.
I took a stab at it with #429 . And it seems to introduce the expected behavior. If you have any suggestions on doing it differently, I look forward to hearing the same and help out if I can.