ros2/launch_ros #442: Dead code line using `lstrip()` without assignment

christophebedard, 2025-02-23:
I found a line that doesn't do anything: https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/actions/node.py#L399

However, `self.__expanded_node_name` is later used in a call to `prefix_namespace(()`: https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/actions/node.py#L423-L424

And, according to `prefix_namespace()`, there's definitely a difference between `self.__expanded_node_name` with and without a leading `/` : https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/utilities/namespace_utils.py#L53-L60

I guess at this point it works fine, in which case the line could just be removed.

k3rnel-pan1c-a, 2025-03-08:
Hi, I'm a first time contributor, is it okay if I start working on this?

christophebedard, 2025-03-08:
Of course! Let us know if you have any questions.

kscottz, 2025-04-01:
@k3rnel-pan1c-a how is it going? Need some help?

TroublemaKer-6, 2025-04-08:
hi, I have created a pr for this issue, hope it would help

christophebedard, 2025-04-08:
@k3rnel-pan1c-a had said they wanted to work on this, but we haven't heard back from them, so I think that's fine. Thank you!

christophebedard, 2025-04-18:
Fixed by #456

closed 2025-04-18 by christophebedard