ros2/launch_ros #443: Update parameters typing to leverage recursive types

christophebedard, 2025-02-24:
Looks like recursive types are supported by mypy now: https://github.com/python/mypy/issues/731.

1. There's a TODO here: https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/parameters_type.py#L55-L57
    - Update the code to leverage recursive types and remove the TODO.
2. There's also a comment below that mentions "(flattened to avoid having a recursive type)," which could therefore probably also be updated, although I'm not quite sure what to do there: https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/parameters_type.py#L72-L73
    - I'm not exactly sure what it refers to. However, for this one, we could probably remove "(flattened to avoid having a recursive type)" from the comment and leave the code as-is.
3. Finally, there are a few references to "making/appeasing mypy" in [`launch_ros/utilities/normalize_parameters.py`](https://github.com/ros2/launch_ros/blob/3569f0d76c68884ca8796c0134bad29e8732d868/launch_ros/launch_ros/utilities/normalize_parameters.py). Perhaps that could be revisited to see if we can simplify it. `# type:` comments could also be updated to the more modern type annotations.

InvincibleRMC, 2025-02-24:
`rhel`  mypy's version is 0.942. So Using the recursive types would fail since version >= 1.7 is needed.

christophebedard, 2025-02-24:
Good point! But I believe it's using 0.971.

* On RHEL 9 ([target for Jazzy and Rolling](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)), we install `python3-mypy`, which seems to be version 0.971: https://pkgs.org/search/?q=python3-mypy. See docs: https://docs.ros.org/en/rolling/Installation/RHEL-Install-RPMs.html#install-development-tools-optional. See also ROS 2 CI: https://github.com/ros2/ci/blob/022da8a9eaa83b4ae8f7b770bdd2a4c700bf1d70/linux_docker_resources/Dockerfile-RHEL#L141
* On RHEL 8 ([target for Humble](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027)), we install `mypy` from `pip`. In the docs, we install version 0.931 specifically: https://docs.ros.org/en/humble/Installation/Alternatives/RHEL-Development-Setup.html#install-development-tools-and-ros-tools. In ROS 2 CI, we install version 0.942: https://github.com/ros2/ci/blob/022da8a9eaa83b4ae8f7b770bdd2a4c700bf1d70/linux_docker_resources/Dockerfile-RHEL#L202-L207.

I guess we'll have to wait a little while more then, since I don't think we'll switch to using `pip` to install a newer `mypy` version. I'll just leave this open for now.

alsora, 2025-03-06:
Hi @christophebedard, we discussed this issue in the waffle meeting and we decided to re-add "help wanted" label.
Do you have any thoughts on that?

christophebedard, 2025-03-10:
I removed it because this issue wasn't immediately actionable based on my quick investigation, but, sure, that works for me!