ros2/launch_ros #445: LifecycleNode autostart=True fails when running the same executable twice with different names

tonynajjar, 2025-03-07:
```
import os
import sys

import launch  # noqa: E402
import launch.actions  # noqa: E402
import launch.events  # noqa: E402

import launch_ros.actions  # noqa: E402
import launch_ros.events  # noqa: E402
import launch_ros.events.lifecycle  # noqa: E402

def main(argv=sys.argv[1:]):
    ld = launch.LaunchDescription()
    talker_node = launch_ros.actions.LifecycleNode(
        name='talker',
        namespace='',
        package='lifecycle',
        executable='lifecycle_talker',
        output='screen',
        emulate_tty=True,
        autostart=True)
    talker2_node = launch_ros.actions.LifecycleNode(
        name='talker2',
        namespace='',
        package='lifecycle',
        executable='lifecycle_talker',
        output='screen',
        emulate_tty=True,
        autostart=True)
    ld.add_action(talker_node)
    ld.add_action(talker2_node)
    ls = launch.LaunchService(argv=argv)
    ls.include_launch_description(ld)
    return ls.run()

if __name__ == '__main__':
    main()
```

run this snippet enough times (10 should be enough) and you'll get an error

```
INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_talker-1]: process started with pid [1311355]
[INFO] [lifecycle_talker-2]: process started with pid [1311356]
[ERROR] [launch_ros.utilities.lifecycle_event_manager]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/talker2'
[lifecycle_talker-2] [WARN] [1741364109.137012849] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[lifecycle_talker-2] [ERROR] [1741364109.137131130] [talker2]: Unable to start transition 3 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_talker-1] [INFO] [1741364109.138277845] [talker]: on_configure() is called.
[lifecycle_talker-1] [INFO] [1741364109.138492406] [talker]: on_activate() is called.
[lifecycle_talker-2] [INFO] [1741364109.138827753] [talker2]: on_configure() is called.
```

FYI @SteveMacenski as author of the feature (thanks btw)

SteveMacenski, 2025-03-07:
I'm not sure its related to my contribution if its flaky; I would think if there was a problem that it would deterministically fail each time. I looked over the PR again in detail, and I don't see anywhere the matchers could be wrong to select the wrong node [except maybe here](https://github.com/ros2/launch_ros/blob/rolling/launch_ros/launch_ros/event_handlers/on_state_transition.py#L73), but I don't think that's actually likely the issue. Further, trying to transition to ACTIVATE before either of them are configured doesn't make me think that the issue is matcher related anyway since neither would be able to do that yet.

I saw things like that when testing when I was control+Cing things very fast and frequently when there were still instances of old nodes running that were orphaned from previous tests, but never on a clean system where I validated that there were no background nodes still running. 

Are you sure this has to do with multiple nodes of the same type, or just two nodes coming up when doing alot of up-down testing and the traffic that multiple nodes provide?

tonynajjar, 2025-03-07:
I remember testing different executables and it did not reproduce. Also i dont think it's related to a cleanup issue as I've experienced the issue in a freshly spun-up socker container. FYI for my application I was launching 2 nav2 velocity smoother nodes

SteveMacenski, 2025-03-14:
@tonynajjar got some time this afternoon to take a look, see solution: https://github.com/ros2/launch_ros/pull/449

tonynajjar, 2025-03-25:
All PRs are merged, will close this issue

closed 2025-03-25 by tonynajjar