ros2/launch_ros #448: Option to Launch ROS Node Parameters from YAML with Optional Namespace

meetgandhi-dev, 2025-03-12:
Hi,

Thank you for your excellent work!

We’ve started utilizing ROS namespaces for our nodes as part of a specific requirement. To pass the namespace to all the nodes in the launch file, we’re using the `PushROSNamespace` action. This setup requires the namespace to be enabled only if a certain environment variable is set, which we’ve managed to accomplish using a `conditional If` argument, and it’s working as expected.

However, we’re encountering some difficulties with managing parameters through a YAML file. When using a namespace, the YAML file should have the namespace as the first key, but when not using a namespace, it should not include it. Each robot has a unique namespace that is determined at boot time, and we prefer not to create multiple variants of YAML files for each case.

Is there an option to provide an optional namespace when specifying parameter files?

I came across a similar solution in the Nav2 bringup launch file ([here](https://github.com/ros-navigation/navigation2/blob/6ef3d7bfd0f617bd0751cdbe7a58c3f9f66fe882/nav2_bringup/launch/bringup_launch.py#L57C1-L73C1)), which addresses this requirement. Does the current module support something like this? If not, do you think adding this feature would be beneficial?

christophebedard, 2025-03-31:
This is an interesting feature request. As far as I know, it is not currently natively supported; PRs are definitely welcome!

I think Nav2's solution (using parameters file as a template) is probably more of a workaround. I haven't looked into it too much, but I wonder if/why using substitutions in their params file doesn't work.

kscottz, 2025-04-01:
@christophebedard do you think this is a good candidate for community support?

christophebedard, 2025-04-01:
I think this one requires a bit of investigation just to see if substitutions could somehow be used in the parameters files. It sounds like @meetgandhi-dev might be open to that. Otherwise, yeah we can ask someone else from the community.

meetgandhi-dev, 2025-04-11:
> I think this one requires a bit of investigation just to see if substitutions could somehow be used in the parameters files. It sounds like [@meetgandhi-dev](https://github.com/meetgandhi-dev) might be open to that. Otherwise, yeah we can ask someone else from the community.

@christophebedard Yes. I would be happy to help on this.

meetgandhi-dev, 2025-04-11:
> [@meetgandhi-dev](https://github.com/meetgandhi-dev) we might try to get the community to contribute this feature. This contributor might be fairly new to ROS 2 launch development. What might help make that happen is a couple of concrete examples (i.e. the state of the world with name spaces and without name spaces). Would you be willing to help test and review the resulting code?

@kscottz sure. Let me upload a stub with all the details.

KoksalBerkay, 2026-09-23:
Hi @christophebedard and @kscottz — I saw the earlier discussion where a stub was planned, but couldn’t find a follow-up PR. Is anyone still working on this? If not, I’d be happy to investigate whether launch substitutions can handle the namespaced and unnamespaced cases.

kscottz, 2026-10-08:
@KoksalBerkay I don't think the stubs ever got created. 

If you want to try and build this feature you are more than welcome to give it a try.