ros2/launch_ros #460: LoadComposableNodes does not parse parameters from files correctly

huweiATgithub, 2025-04-14:
### Operating System:

```Ubuntu 24.04 under Docker on Windows WSL2```

### ROS version or commit hash:

jazzy

### Steps to reproduce issue

# step to reproduce

1. a `ComposableNode` action
    ```python
    ComposableNode(
        package="pkg",
        plugin="NodeCls",
        name="bar",
        namespace="foo",
        parameters=["path-to-params-file"],
    )
2. param file:
   ```yaml
      /*/bar:
        ros__parameters:
          param1: 500
   ```
3. launch the node

### Expected behavior

`param1` being loaded correctly to the Node's overrides

### Actual behavior

the parameters in the param file has not been added to the parameters field of the LoadRequest.
This is due to that **

### Additional information

According to the code https://github.com/ros2/launch_ros/blob/9907300a51aae1188530ea60d88e03ca72c5dbea/launch_ros/launch_ros/utilities/to_parameters_list.py#L86-L101
which is called to take parameters from params_dict parsed from a parameter file.

When matching the node name, only two cases are considered:
- the `**` wildcards
- node name match literally

However, `*` matches a single token** ([ref1](https://design.ros2.org/articles/ros_command_line_arguments.html), [ref2](https://design.ros2.org/articles/static_remapping.html#match-part-of-a-rule)) has not been implemented in the `to_parameters_list` method as quoted above.

This results in different behavior using components vs using `ros2 run`.

## Possible resolution
Implement the node name matching codes parallel to that `rclcpp` (I can't find where the codes are).
Besides, I think we may let the rclcpp api handle the parameter file the same way as `--ros-args --params-file xxx` instead of handling them in the `launch_ros` library.

emrekuru97, 2025-05-06:
Hello, I reproduced the bug on Docker with Ubuntu 24.04 and ROS 2 Jazzy. Should the PR that fixes this be created against the Jazzy branch?

christophebedard, 2025-05-06:
It should target the `rolling` branch. Once that's merged, we can consider backporting it to other branches such as `jazzy` if it makes sense.

bjsowa, 2025-05-21:
Isn't that already fixed in #369 ?

fujitatomoya, 2025-05-21:
actually it did, but 2 years old PR. maybe https://github.com/ros2/launch_ros/pull/465 can take the contents of https://github.com/ros2/launch_ros/pull/369, and close https://github.com/ros2/launch_ros/pull/369

emrekuru97, 2025-05-22:
Took the test part from #369 and incorporated it into #465. Also added a test case for /*. Since I don't have permission, I can't close #369 (just commented),  someone with the right access can do that.

fujitatomoya, 2025-05-29:
@emrekuru97 thanks for your contribution.

emrekuru97, 2025-05-29:
@fujitatomoya Happy to contribute! I noticed that the backport PR's are not passing. Would you like me to take a look at it as well? Thanks!

PR #465 (2025-05-29): Fix: LoadComposableNodes fails to parse wildcard param files correctl…
…y (#460)

## Description

Ported the [function](https://github.com/ros2/rclcpp/blob/f3d66b892e6a7931ac2dbf3d2e5c0c1558312e10/rclcpp/src/rclcpp/parameter_map.cpp#L28-L33) to python. And used it properly for fixing the bug.
Fixes #460 

### Is this user-facing behavior change?

Yes, users will now see parameters like `param1` being correctly loaded from the parameter file when using a `ComposableNode` action.

1. ComposableNode action
```
ComposableNode(
    package="pkg",
    plugin="NodeCls",
    name="bar",
    namespace= [... cut]
PR #466 (2025-05-30): Fix: LoadComposableNodes fails to parse wildcard param files correctl… (backport #465)
…y (#460)

## Description

Ported the [function](https://github.com/ros2/rclcpp/blob/f3d66b892e6a7931ac2dbf3d2e5c0c1558312e10/rclcpp/src/rclcpp/parameter_map.cpp#L28-L33) to python. And used it properly for fixing the bug.
Fixes #460 

### Is this user-facing behavior change?

Yes, users will now see parameters like `param1` being correctly loaded from the parameter file when using a `ComposableNode` action.

1. ComposableNode action
```
ComposableNode(
    package="pkg",
    plugin="NodeCls",
    name="bar" [... cut]